Robot-Will/Stino #432: compile stop at step 2

acs8369, 2017-04-13:
I manual add some libraries, some libraries can compiled at first time but some not. 

[Build] D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
*** compiler still finding source file ***
sublime hang

https://github.com/adafruit/Adafruit-RGB-LCD-Shield-Library

Robot-Will, 2017-04-14:
Fixed. Thanks.

acs8369, 2017-04-14:
fresh install sublime text3 portable, stino dev branch

* first compile *

[Build] D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.3%] Compiling HelloWorld.ino.cpp...
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld/HelloWorld.ino:12:35: fatal error: Adafruit_RGBLCDShield.h: No such file or directory
 #include <Adafruit_RGBLCDShield.h>
                                   ^
compilation terminated.
[Build] Error occurred.
-----------------------------------------------------------------------------------------------------
** second time compile **

[Build] D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.2%] Compiling HelloWorld.ino.cpp...
[6.5%] Compiling Wire.cpp...
[9.7%] Compiling Adafruit_RGBLCDShield.cpp...
[12.9%] Compiling twi.c...
[16.1%] Compiling abi.cpp...
[19.4%] Compiling CDC.cpp...
[22.6%] Compiling HardwareSerial.cpp...
[25.8%] Compiling HardwareSerial0.cpp...
[29.0%] Compiling HardwareSerial1.cpp...
[32.3%] Compiling HardwareSerial2.cpp...
[35.5%] Compiling HardwareSerial3.cpp...
[38.7%] Compiling IPAddress.cpp...
[41.9%] Compiling main.cpp...
[45.2%] Compiling new.cpp...
[48.4%] Compiling PluggableUSB.cpp...
[51.6%] Compiling Print.cpp...
[54.8%] Compiling Stream.cpp...
[58.1%] Compiling Tone.cpp...
[61.3%] Compiling USBCore.cpp...
[64.5%] Compiling WMath.cpp...
[67.7%] Compiling WString.cpp...
[71.0%] Compiling hooks.c...
[74.2%] Compiling WInterrupts.c...
[77.4%] Compiling wiring.c...
[80.6%] Compiling wiring_analog.c...
[83.9%] Compiling wiring_digital.c...
[87.1%] Compiling wiring_pulse.c...
[90.3%] Compiling wiring_shift.c...
[93.5%] Compiling wiring_pulse.S...
[96.8%] Linking everything together...
[100.0%] Creating binary files...
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `Adafruit_RGBLCDShield::_digitalWrite(unsigned char, unsigned char)':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:333: undefined reference to `Adafruit_MCP23017::digitalWrite(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `Adafruit_RGBLCDShield::write4bits(unsigned char)':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:389: undefined reference to `Adafruit_MCP23017::readGPIOAB()'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:400: undefined reference to `Adafruit_MCP23017::writeGPIOAB(unsigned int)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:405: undefined reference to `Adafruit_MCP23017::writeGPIOAB(unsigned int)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:408: undefined reference to `Adafruit_MCP23017::writeGPIOAB(unsigned int)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `_pinMode':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:352: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:352: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `begin':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:119: undefined reference to `Adafruit_MCP23017::begin()'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:121: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:122: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:123: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:127: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:129: undefined reference to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o:D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:130: more undefined references to `Adafruit_MCP23017::pinMode(unsigned char, unsigned char)' follow
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `begin':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:136: undefined reference to `Adafruit_MCP23017::pullUp(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans0.ltrans.o: In function `readButtons':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:433: undefined reference to `Adafruit_MCP23017::digitalRead(unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccaOIrMa.ltrans1.ltrans.o: In function `Adafruit_RGBLCDShield::setBacklight(unsigned char) [clone .constprop.31]':
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:343: undefined reference to `Adafruit_MCP23017::digitalWrite(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:344: undefined reference to `Adafruit_MCP23017::digitalWrite(unsigned char, unsigned char)'
D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/Adafruit_RGBLCDShield.cpp:345: undefined reference to `Adafruit_MCP23017::digitalWrite(unsigned char, unsigned char)'
collect2.exe: error: ld returned 1 exit status
[Build] Error occurred.

Robot-Will, 2017-04-14:
I can not repeat your error. Arduino Uno and Mega2560.

```
[Build] D:/Documents/Arduino/libraries/Adafruit-RGB-LCD-Shield-Library/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.1%] Compiling HelloWorld.ino.cpp...
[6.2%] Compiling Wire.cpp...
[9.4%] Compiling Adafruit_RGBLCDShield.cpp...
[12.5%] Compiling Adafruit_MCP23017.cpp...
[15.6%] Compiling twi.c...
[18.8%] Compiling abi.cpp...
[21.9%] Compiling CDC.cpp...
[25.0%] Compiling HardwareSerial.cpp...
[28.1%] Compiling HardwareSerial0.cpp...
[31.2%] Compiling HardwareSerial1.cpp...
[34.4%] Compiling HardwareSerial2.cpp...
[37.5%] Compiling HardwareSerial3.cpp...
[40.6%] Compiling IPAddress.cpp...
[43.8%] Compiling main.cpp...
[46.9%] Compiling new.cpp...
[50.0%] Compiling PluggableUSB.cpp...
[53.1%] Compiling Print.cpp...
[56.2%] Compiling Stream.cpp...
[59.4%] Compiling Tone.cpp...
[62.5%] Compiling USBCore.cpp...
[65.6%] Compiling WMath.cpp...
[68.8%] Compiling WString.cpp...
[71.9%] Compiling hooks.c...
[75.0%] Compiling WInterrupts.c...
[78.1%] Compiling wiring.c...
[81.2%] Compiling wiring_analog.c...
[84.4%] Compiling wiring_digital.c...
[87.5%] Compiling wiring_pulse.c...
[90.6%] Compiling wiring_shift.c...
[93.8%] Compiling wiring_pulse.S...
[96.9%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 5,938 bytes (18.4%) of program storage space. Maximum is 32,256 bytes.
Global variables use 492 bytes (24.0%) of dynamic memory, leaving 1,556 bytes for local variables. Maximum is 2,048 bytes.
[Build done.]
```

acs8369, 2017-04-14:
test on Uno and Mega2560 or Leonardo give nearly same result
 
[Build] D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld...
// include the library code:
#include <Wire.h>
#include <Adafruit_RGBLCDShield.h>
#include <utility/Adafruit_MCP23017.h> // may be this line is problem
------------------------------------------------------------------------------------------------------
> I can not repeat your error. Arduino Uno and Mega2560.

[Build] D:/Documents/Arduino/libraries/Adafruit-RGB-LCD-Shield-Library/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.1%] Compiling HelloWorld.ino.cpp...
[6.2%] Compiling Wire.cpp...
[9.4%] Compiling Adafruit_RGBLCDShield.cpp...
[12.5%] Compiling Adafruit_MCP23017.cpp...
[15.6%] Compiling twi.c...
--------------------------------------------------------------------------------------------
***in my compiled output missing this line [12.5%] Compiling Adafruit_MCP23017.cpp...
[Build] D:/DEV/Arduino/libraries/Adafruit_RGBLCDShield/examples/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.2%] Compiling HelloWorld.ino.cpp...
[6.5%] Compiling Wire.cpp...
[9.7%] Compiling Adafruit_RGBLCDShield.cpp...
[12.9%] Compiling twi.c...

Robot-Will, 2017-04-14:
So? ?

You may check the Adafruit-RGB-LCD-Shield-Library you installed.

acs8369, 2017-04-14:
this is result from default sd libraries 

[Build] D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.2%] Compiling CardInfo.ino.cpp...
[6.5%] Compiling SPI.cpp...
[9.7%] Compiling File.cpp...
[12.9%] Compiling SD.cpp...
[16.1%] Compiling abi.cpp...
[19.4%] Compiling CDC.cpp...
[22.6%] Compiling HardwareSerial.cpp...
[25.8%] Compiling HardwareSerial0.cpp...
[29.0%] Compiling HardwareSerial1.cpp...
[32.3%] Compiling HardwareSerial2.cpp...
[35.5%] Compiling HardwareSerial3.cpp...
[38.7%] Compiling IPAddress.cpp...
[41.9%] Compiling main.cpp...
[45.2%] Compiling new.cpp...
[48.4%] Compiling PluggableUSB.cpp...
[51.6%] Compiling Print.cpp...
[54.8%] Compiling Stream.cpp...
[58.1%] Compiling Tone.cpp...
[61.3%] Compiling USBCore.cpp...
[64.5%] Compiling WMath.cpp...
[67.7%] Compiling WString.cpp...
[71.0%] Compiling hooks.c...
[74.2%] Compiling WInterrupts.c...
[77.4%] Compiling wiring.c...
[80.6%] Compiling wiring_analog.c...
[83.9%] Compiling wiring_digital.c...
[87.1%] Compiling wiring_pulse.c...
[90.3%] Compiling wiring_shift.c...
[93.5%] Compiling wiring_pulse.S...
[96.8%] Linking everything together...
[100.0%] Creating binary files...
C:\Users\Nattawut\AppData\Local\Temp\cc8zKqLy.ltrans0.ltrans.o: In function `SdFile::SdFile()':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:140: undefined reference to `vtable for SdFile'
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:140: undefined reference to `vtable for SdFile'
C:\Users\Nattawut\AppData\Local\Temp\cc8zKqLy.ltrans0.ltrans.o: In function `setup':
D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo/CardInfo.ino:50: undefined reference to `Sd2Card::init(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\cc8zKqLy.ltrans0.ltrans.o: In function `init':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:454: undefined reference to `SdVolume::init(Sd2Card*, unsigned char)'
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:454: undefined reference to `SdVolume::init(Sd2Card*, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\cc8zKqLy.ltrans0.ltrans.o: In function `openRoot':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:349: undefined reference to `SdFile::openRoot(SdVolume*)'
C:\Users\Nattawut\AppData\Local\Temp\cc8zKqLy.ltrans0.ltrans.o: In function `setup':
D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo/CardInfo.ino:106: undefined reference to `SdFile::ls(unsigned char, unsigned char)'
collect2.exe: error: ld returned 1 exit status
[Build] Error occurred.

--------------------------------------------

Robot-Will, 2017-04-14:
It's weird. Do you update this plugin to the newest version?

The difference is
```
[14.7%] Compiling Sd2Card.cpp...
[17.6%] Compiling SdFile.cpp...
[20.6%] Compiling SdVolume.cpp...
```

```
[Build] E:/dev/arduino/arduino-1.8.2/libraries/SD/examples/CardInfo...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[2.9%] Compiling CardInfo.ino.cpp...
[5.9%] Compiling SPI.cpp...
[8.8%] Compiling File.cpp...
[11.8%] Compiling SD.cpp...
[14.7%] Compiling Sd2Card.cpp...
[17.6%] Compiling SdFile.cpp...
[20.6%] Compiling SdVolume.cpp...
[23.5%] Compiling abi.cpp...
[26.5%] Compiling CDC.cpp...
[29.4%] Compiling HardwareSerial.cpp...
[32.4%] Compiling HardwareSerial0.cpp...
[35.3%] Compiling HardwareSerial1.cpp...
[38.2%] Compiling HardwareSerial2.cpp...
[41.2%] Compiling HardwareSerial3.cpp...
[44.1%] Compiling IPAddress.cpp...
[47.1%] Compiling main.cpp...
[50.0%] Compiling new.cpp...
[52.9%] Compiling PluggableUSB.cpp...
[55.9%] Compiling Print.cpp...
[58.8%] Compiling Stream.cpp...
[61.8%] Compiling Tone.cpp...
[64.7%] Compiling USBCore.cpp...
[67.6%] Compiling WMath.cpp...
[70.6%] Compiling WString.cpp...
[73.5%] Compiling hooks.c...
[76.5%] Compiling WInterrupts.c...
[79.4%] Compiling wiring.c...
[82.4%] Compiling wiring_analog.c...
[85.3%] Compiling wiring_digital.c...
[88.2%] Compiling wiring_pulse.c...
[91.2%] Compiling wiring_shift.c...
[94.1%] Compiling wiring_pulse.S...
[97.1%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 9,184 bytes (28.5%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,269 bytes (62.0%) of dynamic memory, leaving 779 bytes for local variables. Maximum is 2,048 bytes.
[Build done.]
```

acs8369, 2017-04-14:
I use newest version Stino-Stino-Dev from Branch: Stino-Dev

fresh install sublimetext3 portable (delete old sublime portable folder and arduino15 before reinstall)
result is 

---------------------------------------------------------------------------
first compile

[Build] D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.3%] Compiling CardInfo.ino.cpp...
D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo/CardInfo.ino:25:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
[Build] Error occurred.
--------------------------------------------------------------------------
second compile

[Build] D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.2%] Compiling CardInfo.ino.cpp...
[6.5%] Compiling File.cpp...
[9.7%] Compiling SD.cpp...
[12.9%] Compiling SPI.cpp...
[16.1%] Compiling abi.cpp...
[19.4%] Compiling CDC.cpp...
[22.6%] Compiling HardwareSerial.cpp...
[25.8%] Compiling HardwareSerial0.cpp...
[29.0%] Compiling HardwareSerial1.cpp...
[32.3%] Compiling HardwareSerial2.cpp...
[35.5%] Compiling HardwareSerial3.cpp...
[38.7%] Compiling IPAddress.cpp...
[41.9%] Compiling main.cpp...
[45.2%] Compiling new.cpp...
[48.4%] Compiling PluggableUSB.cpp...
[51.6%] Compiling Print.cpp...
[54.8%] Compiling Stream.cpp...
[58.1%] Compiling Tone.cpp...
[61.3%] Compiling USBCore.cpp...
[64.5%] Compiling WMath.cpp...
[67.7%] Compiling WString.cpp...
[71.0%] Compiling hooks.c...
[74.2%] Compiling WInterrupts.c...
[77.4%] Compiling wiring.c...
[80.6%] Compiling wiring_analog.c...
[83.9%] Compiling wiring_digital.c...
[87.1%] Compiling wiring_pulse.c...
[90.3%] Compiling wiring_shift.c...
[93.5%] Compiling wiring_pulse.S...
[96.8%] Linking everything together...
[100.0%] Creating binary files...
C:\Users\Nattawut\AppData\Local\Temp\ccm9JLiQ.ltrans0.ltrans.o: In function `SdFile::SdFile()':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:140: undefined reference to `vtable for SdFile'
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:140: undefined reference to `vtable for SdFile'
C:\Users\Nattawut\AppData\Local\Temp\ccm9JLiQ.ltrans0.ltrans.o: In function `setup':
D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo/CardInfo.ino:52: undefined reference to `Sd2Card::init(unsigned char, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccm9JLiQ.ltrans0.ltrans.o: In function `init':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:454: undefined reference to `SdVolume::init(Sd2Card*, unsigned char)'
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:454: undefined reference to `SdVolume::init(Sd2Card*, unsigned char)'
C:\Users\Nattawut\AppData\Local\Temp\ccm9JLiQ.ltrans0.ltrans.o: In function `openRoot':
D:/Software/arduino-1.8.2/libraries/SD/src/utility/SdFat.h:349: undefined reference to `SdFile::openRoot(SdVolume*)'
C:\Users\Nattawut\AppData\Local\Temp\ccm9JLiQ.ltrans0.ltrans.o: In function `setup':
D:/Software/arduino-1.8.2/libraries/SD/examples/CardInfo/CardInfo.ino:108: undefined reference to `SdFile::ls(unsigned char, unsigned char)'
collect2.exe: error: ld returned 1 exit status
[Build] Error occurred.

alexrv42, 2017-08-06:
Have the same error when building sketches. Any solutions?

> 
[Build] C:/Users/user/Documents/Arduino/as...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.6%] Compiling as.ino.cpp...
[7.1%] Compiling abi.cpp...
[10.7%] Compiling CDC.cpp...
[14.3%] Compiling HardwareSerial.cpp...
[17.9%] Compiling HardwareSerial0.cpp...
[21.4%] Compiling HardwareSerial1.cpp...
[25.0%] Compiling HardwareSerial2.cpp...
[28.6%] Compiling HardwareSerial3.cpp...
[32.1%] Compiling IPAddress.cpp...
[35.7%] Compiling main.cpp...
[39.3%] Compiling new.cpp...
[42.9%] Compiling PluggableUSB.cpp...
[46.4%] Compiling Print.cpp...
[50.0%] Compiling Stream.cpp...
[53.6%] Compiling Tone.cpp...
[57.1%] Compiling USBCore.cpp...
[60.7%] Compiling WMath.cpp...
[64.3%] Compiling WString.cpp...
[67.9%] Compiling hooks.c...
[71.4%] Compiling WInterrupts.c...
[75.0%] Compiling wiring.c...
[78.6%] Compiling wiring_analog.c...
[82.1%] Compiling wiring_digital.c...
[85.7%] Compiling wiring_pulse.c...
[89.3%] Compiling wiring_shift.c...
[92.9%] Compiling wiring_pulse.S...
[96.4%] Linking everything together...
[100.0%] Creating binary files...
collect2.exe: fatal error: cannot find 'ld'
compilation terminated.
[Build] Error occurred.

acs8369, 2017-08-07:
@alexrv42 

I already solve compile error problem before I post this issue. 
problem: compiler cannot find "include header file (xxx.h)" in sketch folder, platform folder and arduino application folder
solution: manual add file or library even if it already exit. After the first time resolve, the second time it will automatically find it's dependency.
ex.
- menu -> Arduino -> Import Library -> library xxxxx
- menu -> Arduino -> Install Library -> library xxxxx

acs8369, 2017-08-08:
@alexrv42 <https://github.com/alexrv42>

I already solve compile error problem before I post this issue.
problem: compiler cannot find "include header file (xxx.h)" in sketch
folder, platform folder and arduino application folder
solution: manual add file or library even if it already exit. After the
first time resolve, the second time it will automatically find it's
dependency.
ex.

   - menu -> Arduino -> Import Library -> library xxxxx
   - menu -> Arduino -> Install Library -> library xxxxx