Robot-Will/Stino #433: compile stop at step 1

hunandy14, 2017-04-14:
[Build] C:/Users/Charlotte/Desktop/Blink...
[Step 1] Check Toolchain.

stop here.
i tyr reinstall. can't fix.

i upload to uno, up to com1.
but com1 is not correct,

i wait long time. 
then i close "sublime text".
then it stop at step1.   QuQ...

Robot-Will, 2017-04-14:
Ctrl+` to show the error messages. Thanks.

hunandy14, 2017-04-14:
－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
DPI scale: 1
startup, version: 3126 windows x64 channel: stable
executable: /Q/Program Files/Sublime Text 3/sublime_text.exe
working dir: /Q/Program Files/Sublime Text 3
packages path: /Q/Program Files/Sublime Text 3/Data/Packages
state path: /Q/Program Files/Sublime Text 3/Data/Local
zip path: /Q/Program Files/Sublime Text 3/Packages
zip path: /Q/Program Files/Sublime Text 3/Data/Installed Packages
ignored_packages: ["Markdown", "Vintage"]
pre session restore time: 0.415389
startup time: 0.469389
first paint time: 0.471389
first paint time: 0.471389
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.02-bz2
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pytz
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin AdvancedNewFile.AdvancedNewFile
reloading plugin ApplySyntax.ApplySyntax
reloading plugin ApplySyntax.changes
[... 74 lines cut ...]
OmniMarkupPreviewer: [INFO] Loaded renderer: MediaWikiRenderer
OmniMarkupPreviewer: [INFO] Loaded renderer: OrgRenderer
OmniMarkupPreviewer: [INFO] Loaded renderer: PodRenderer
OmniMarkupPreviewer: [INFO] Loaded renderer: RDocRenderer
OmniMarkupPreviewer: [INFO] Loaded renderer: RstRenderer
OmniMarkupPreviewer: [INFO] Loaded renderer: TextileRenderer
WARNING:process:Could not import subprocess32 module, falling back to subprocess module
top level value must be an array
Package Control: Skipping automatic upgrade, last run at 2017-04-14 17:21:59, next run at 2017-04-14 18:21:59 or after
Exception in thread Thread-17:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "Q:\Program Files\Sublime Text 3\Data\Packages\Arduino-like IDE\libs\base_utils\task_queue.py", line 86, in _run
    self._consumer(*args)
  File "Q:\Program Files\Sublime Text 3\Data\Packages\Arduino-like IDE\libs\stino_runtime\__init__.py", line 2054, in build_sketch
    is_ready = check_platform_dep()
  File "Q:\Program Files\Sublime Text 3\Data\Packages\Arduino-like IDE\libs\stino_runtime\__init__.py", line 708, in check_platform_dep
    is_ready = check_tools_deps(build_platform_info)
  File "Q:\Program Files\Sublime Text 3\Data\Packages\Arduino-like IDE\libs\stino_runtime\__init__.py", line 752, in check_tools_deps
    if go_down:
UnboundLocalError: local variable 'go_down' referenced before assignment

－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－
－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－－

hunandy14, 2017-04-14:
Successful fix.
thank you.

closed 2017-04-14 by Robot-Will