Robot-Will/Stino #435: Problem nistalling ESP32 support

KurtE, 2017-04-14:
I wanted to try out some stuff on an ESP32 board.  

So I install support to Arduino for this following the instructions up at: 
https://github.com/espressif/arduino-esp32

I tried it both on Windows 10 and now on a MAC following the instructions at: 
https://github.com/espressif/arduino-esp32#instructions-for-mac

I then loaded up an example program (like their hardware example RepeatTimer).  I choose the ESP32 Dev Module as the board type and Arduino completed the build.

I then tried doing the build using Stino.
I selected the Platform: Sketch book => espressif
I selected the board : ESP32 Dev Module and tried to do a build

It fails on both machines:  Messages on Windows:
```
[Build] C:/Users/kurte/Documents/Arduino/hardware/espressif/esp32/libraries/ESP32/examples/Timer/RepeatTimer...
[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.
```
Not sure if the Console helps: 
```
DPI scale: 1
startup, version: 3126 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.292411
startup time: 0.339411
first paint time: 0.339411
first paint time: 0.339411
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin QuickPanelEnhanced.quick_panel_enhanced
reloading plugin SidebarHoverToggle.sidebar_hover_toggle
reloading plugin Sublimerge 3.Sublimerge
reloading plugin Stino.StinoCommands
plugins loaded
top level value must be an array
Package Control: Skipping automatic upgrade, last run at 2017-04-14 05:31:58, next run at 2017-04-14 06:31:58 or after
```

Robot-Will, 2017-04-14:
Follow the steps to install Esp8266 boards:

1. In this page (https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls) you will find the Esp8266's package index file address:

```
ESP8266 Community: http://arduino.esp8266.com/stable/package_esp8266com_index.json

Generic ESP8266 modules
Olimex MOD-WIFI-ESP8266
NodeMCU 0.9 (ESP-12)
NodeMCU 1.0 (ESP-12E)
Adafruit HUZZAH ESP8266 (ESP-12)
SparkFun Thing
SweetPea ESP-210
WeMos D1
WeMos D1 mini
```

2. Menu -> Install platform -> Add Package

3. Edit packages.stino-settings, add the esp8266 package index file address into this file, and save:
```
{
    "arduino": "http://downloads.arduino.cc/packages/package_index.json",
    "arduino_lib": "http://downloads.arduino.cc/libraries/library_index.json",
    "esp8266": "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
}
```

4. Restart SublimeText, and it will down load the esp8266 package index file. Then menu item `esp8266` will appear in `Menu -> Install Platform`, and click the version you want to install, it will download all.

5. After installation, you will find `esp8266` in `Menu -> Platform`, and you can use it. 

Thanks.

KurtE, 2017-04-14:
Thanks I am trying that now.  Would make a lot more sense. 

So I got to the point of restart sublime test .  The menu item for esp8266 shows sub menus going through sparkfun… I select it and then see:

 

```

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] Download started.

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 12% (0.00 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 24% (0.01 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 36% (0.01 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 48% (0.01 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 59% (0.01 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 71% (0.02 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 83% (0.02 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] 95% (0.02 M / 0.02 M)

[http://arduino.esp8266.com/stable/package_esp8266com_index.json] Download completed.

[https://cdn.sparkfun.com/assets/learn_tutorials/4/9/5/sparkfun-esp8266-2.1.0.tar.bz2] Waiting for download...

```

Which stops there.  I then look at console and see: 

```

https://cdn.sparkfun.com/assets/learn_tutorials/4/9/5/sparkfun-esp8266-2.1.0.tar.bz2 <urlopen error [WinError 10038] An operation was attempted on something that is not a socket>

```

 

This is on the windows 10 machine, will try also on mac

KurtE, 2017-04-14:
Quick update, 

 

Appears to be working better on MAC.  Will try to cleanup on Window machine and try again.

KurtE, 2017-04-15:
Quick FYI - The board I have is the ESP-WROOM-32 chip and not the ESP8266, so the above platform does not work with it.  

I am able to program it using the IDE I mentioned in the first posting.

machinehistories, 2018-05-27:
I too am struggling to get the esp32 (m5stack) working in stino. It works fine in arduino but I can't get a esp32 menu in stino. Please advise.

closed 2022-02-14 by KurtE