Robot-Will/Stino #436: compile error

hunandy14, 2017-04-15:
originally work is normal.
but i keep on compile , than Error.
i not uninstall any soft.

```
[Build] C:/Users/Charlotte/OneDrive/Git Repository/BT_ATC...
[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.
```

how can i do?

Robot-Will, 2017-04-17:
What platform and borad?

davidyu117, 2017-04-17:
This have happened to me too. 
Platform: Ubuntu.
Board: Arduino NANO

hunandy14, 2017-04-17:
Platform: Windows10
Board: Arduino Uno

hunandy14, 2017-04-19:
do you any idea?  QuQ

i try copy sublime_folder to other computer.  it normal work.
and reinstall it. can't fix.

MatiasCettour, 2017-06-29:
This have happened to me too
Platform: Windows 10
Board: Arduino mega 2560

starock, 2017-08-24:
Me too.  macOS 10.12, Arduino nano

Toolchain is not ready. Please build the sketch after the toolchain installation done.

neal-neal, 2017-08-24:
in my case，wait a minute, it will be ok.

starock, 2017-08-24:
I just solved.  Close Arduino IDE, then return Sublime Text 3 , upload. Works fine!

hunandy14, 2017-08-24:
today I solved.
but not close Arduino IED or wait a minute.
i always close Arduino IED and wait enough time.

I guess soft have update.
thank.

fer3ll, 2017-09-01:
Answear is here! Download and install [TOOLCHAIN forAVR](http://www.atmel.com/dyn/resources/prod_documents/avr-toolchain-installer-3.3.0.710-win32.win32.x86.exe)

mai1015, 2017-10-02:
I cannot get my toolchain downloaded from sublime text 3. It was working on the other computer where it automatically install the toolchain.
There is nothing under install platform other then Refresh, Add package, and Add Arduino IDE

System: macOS High Sierra
Board: Arduino UNO

BER7777, 2017-10-17:
#fer3ll, this fix didn't work for me. I installed the file, the problem persists. Do I need to install through Sublime?
Used to work fine until I updated Sublime Text recently.

BER7777, 2017-10-17:
Has anyone figured a fix for this problem or did you all just switched to an alternative soft?

evefos, 2017-11-30:
#BER7777 Issue stil excist i persume, and for alternative soft I would give Atom a try, they have a similar workaround for arduino just like stino.

BER7777, 2017-12-02:
Works for me now. Don't know what has changed, but works as clock.

hunandy14, 2017-12-25:
i get answer!!!!!!

today. i have a new computer.
i install sublimetext3 and arduino1.85 and stino.

but this is not work. i get same error and messeage.
than i change text coding.  change to **utf-8**.

now is work!!!

so 3moth ago, 
I not do anything.
can work.

fer3ll, 2017-12-25:
Everything works, this development environment is the best that I've met. Just have to follow the rules of installation and use.

fer3ll, 2017-12-25:
To perform timer calculations https://www.mikroe.com/timer-calculator