Robot-Will/Stino #437: Sometimes Stino menu gets corrupted.

KurtE, 2017-04-15:
[image]

Every so often I have Stino either not be able to open it's menu or the menu gets corrupted.  I included an image of when it happened just now.   This happened when I switched from one platform type (ESP), to Arduino IDE->Teensy  Often this appears to happen when I have multiple windows open.  For example I have one Arduino sketch window open, plus two Open Folders (actually from open recent)

This is windows 10...  

I wonder at times if it is running out of memory or something of that nature.  
Just in case you might find some useful information, here is the console output, 
```
DPI scale: 1
startup, version: 3126 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/kurte/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.224484
startup time: 0.287484
first paint time: 0.287484
first paint time: 0.287484
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin QuickPanelEnhanced.quick_panel_enhanced
reloading plugin SidebarHoverToggle.sidebar_hover_toggle
reloading plugin Sublimerge 3.Sublimerge
reloading plugin Stino.StinoCommands
plugins loaded
Package Control: No updated packages
reloading /C/Users/kurte/AppData/Roaming/Sublime Text 3/Packages/User/Stino/app_dir.stino-settings
reloading /C/Users/kurte/Documents/Arduino/libraries/ili9341_t3n/ILI9341_t3n.cpp
reloading /C/Users/kurte/Documents/Arduino/libraries/ili9341_t3n/ILI9341_t3n.h
reloading /C/Users/kurte/Documents/Arduino/libraries/ili9341_t3n/ILI9341_t3n.cpp
reloading /C/Users/kurte/Documents/Arduino/libraries/ili9341_t3n/ILI9341_t3n.h
```
Note: the problem clears up if I exit sublimetext and then restart

Robot-Will, 2017-04-17:
Sublime Text creates menu by loading a lot of Main.sublime-menu files, and this soft creates several Main.sublime-menu files to change the contents of the menu. When the files changed, ST will reload all the menu files to create a new menu, so I found ST will not do it right each time, and you see the problem. I can do noting for the problem. Thanks.

KurtE, 2017-04-23:
Thanks, 

But at times what I am wondering is if maybe either memory is getting corrupted or maybe for example there is either memory leaking and/or things like file handles or the like. 

However please note, I am in no way a Python expert.  The most I have done was to earlier figure out how to do a few minor things to get the Teensy limping along with other older system or on some Linux boxes I have modified a ROS module or two that were written in Python. 

When I did a search for leaking file handles and the like, it became very unclear if the systems garbage collection system would automatically close it when all of the handles to it were out of scope.  Some of the websites suggested it is up to each implementation. 

Currently I am getting into the state above reasonably often and/or sublimetext crashes. 

Thanks again

closed 2022-02-14 by KurtE