Robot-Will/Stino #438: Choosing the correct library.

KurtE, 2017-04-16:
Currently I am playing around with the SPI library, So I have a version of the SPI library in my <sketch folder>/libraries directory.  So currently when I build for the Teensy or AVR boards, it uses this one instead of the one that is installed as part of the platform.  For example it uses the on in my sketch folder and not the one in <arduino install>/hardware/teensy/avr/libraries/SPI
Note: My version of the SPI library (sketch folder) has a file library.properties which looks like:
```
name=SPI
version=1.0
author=Arduino
maintainer=Paul Stoffregen
sentence=Enables the communication with devices that use the Serial Peripheral Interface (SPI) Bus.  
paragraph=SPI is a synchronous serial data protocol used by microcontrollers for communicating with one or more peripheral devices quickly over short distances. It uses three lines common to all devices (MISO, MOSI and SCK) and one specific for each device.
category=Communication
url=http://www.arduino.cc/en/Reference/SPI
architectures=avr
includes=SPI.h
```
However right now I am also hacking up the Arduino Due version which is installed in the Arduino15 folders.  

When I compile on using Arduino (1.8.2), it uses the version in the hardware folder as it's architecture is not avr.  
 
As you can see it it's compile output: 
```
Multiple libraries were found for "SPI.h"
 Used: C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\libraries\SPI
 Not used: C:\Users\kurte\Documents\Arduino\libraries\SPI
```

But if I try to build with Stino, it uses the SPI library in the sketch folder

Robot-Will, 2017-04-17:
Yes, I think this is a big problem of Arduino Framework. As at the beginning of Arduino, there is only one  mcu type, that is AVR, so all libs without library.properties work. And then SAM appears and a lot of other mcus were added to the Arduino Framwork, it is difficult to manage the libs because you can not easily tell what lib is suitable for the given mcu. library.properties is a way to get the right lib, but not all libs have this file and follow the Arduino IDE 1.5: Library specification(https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification). 

I do not use the library.properties file and find the libs according to the order of:
1. sketch folder
2. platform folder
3. arduino app folder

I will change the order to:
1. platform folder
2. sketch folder
3. arduino app folder

Thanks.

KurtE, 2017-04-17:
Actually you are maybe better off not changing the order.  Hard to say

KurtE, 2017-04-20:
Actually the new change sort of confused me for awhile as all of a sudden my test app was not using my new code anymore... 

Again hard to say what to do.  I would probably revert to older behavior or make the order selectable...

closed 2022-02-14 by KurtE