Robot-Will/Stino #444: ROS serial compilation failure

normanchen1991, 2017-05-09:
STINO is a fantastic plug-in and thanks so much!
However, it doesn't compile the example code of rosserial_arduino [here]
(http://wiki.ros.org/rosserial_arduino/Tutorials/Hello%20World) after I finished the setup procedures,
and the error messages show that a lot of functions are "not declared in this scope".

Meanwhile, the Arduino IDE (1.8.2) compiles without any problem.

I guess it might have something to do with the custom libraries. Could you please help me with this issue? Thanks so much!

mahyuddin, 2017-05-17:
I'm facing the same problem with rosserial_arduino (on Arduino Uno and ESP8266).

mahyuddin, 2017-05-17:
> [Build] C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling HelloWorld.ino.cpp...
In file included from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/WString.h:27:0,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/IPAddress.h:24,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld/HelloWorld.ino:11:
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/std_msgs/string.h: In member function 'virtual int std_msgs::String::serialize(unsigned char*) const':
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/std_msgs/string.h:26:47: error: 'strlen' was not declared in this scope
       uint32_t length_data = strlen(this->data);
                                               ^
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/std_msgs/string.h:29:57: error: 'memcpy' was not declared in this scope
       memcpy(outbuffer + offset, this->data, length_data);
                                                         ^
In file included from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Stream.h:26:0,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/HardwareSerial.h:31,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:245,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiClient.h:25,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
                 from C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld/HelloWorld.ino:11:
C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.h: In member function 'size_t Print::write(const char*)':
C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Print.h:59:59: error: 'strlen' was not declared in this scope
             return write((const uint8_t *) str, strlen(str));
                                                           ^
In file included from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Updater.h:6:0,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:247,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiClient.h:25,
                 from C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
                 from C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld/HelloWorld.ino:11:
C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/MD5Builder.h: In member function 'void MD5Builder::add(const char*)':
C:/Users/mahyuddin/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/MD5Builder.h:35:65: error: 'strlen' was not declared in this scope
     void add(const char * data){ add((uint8_t*)data, strlen(data)); }
                                                                 ^
In file included from C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros/node_handle.h:41:0,
                 from C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros.h:38,
                 from C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld/HelloWorld.ino:12:
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/rosserial_msgs/TopicInfo.h: In member function 'virtual int rosserial_msgs::TopicInfo::serialize(unsigned char*) const':
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/rosserial_msgs/TopicInfo.h:49:59: error: 'strlen' was not declared in this scope
       uint32_t length_topic_name = strlen(this->topic_name);
                                                           ^
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/rosserial_msgs/TopicInfo.h:52:69: error: 'memcpy' was not declared in this scope
       memcpy(outbuffer + offset, this->topic_name, length_topic_name);
                                                                     ^
In file included from C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros/node_handle.h:42:0,
                 from C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros.h:38,
                 from C:/Users/mahyuddin/Workspace/rosserial_examples/Esp8266/HelloWorld/HelloWorld.ino:12:
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/rosserial_msgs/Log.h: In member function 'virtual int rosserial_msgs::Log::serialize(unsigned char*) const':
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/rosserial_msgs/Log.h:36:45: error: 'strlen' was not declared in this scope
       uint32_t length_msg = strlen(this->msg);
                                             ^
[... cut ...]
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros/node_handle.h:540:56: error: there are no arguments to 'strcpy' that depend on a template parameter, so a declaration of 'strcpy' must be available [-fpermissive]
               strcpy(param[i],req_param_resp.strings[i]);
                                                        ^
C:/Users/mahyuddin/Documents/Arduino/libraries/ros_lib/ros/node_handle.h:540:56: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
[Build] Error occurred.