Robot-Will/Stino #445: ESP8266 WiFi not able to Build

elbrinke, 2017-05-12:
Hi, 

I am using ESP8266 version 2.3.0 library inside of Stino. When I try t build an ESP WiFi project, I get the following errors: 

`
[Build] /Users/eddy/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/examples/WiFiClientEvents...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling WiFiClientEvents.ino.cpp...
[2.9%] Compiling ESP8266WiFi.cpp...
[4.3%] Compiling ESP8266WiFiAP.cpp...
[5.7%] Compiling ESP8266WiFiGeneric.cpp...
[7.1%] Compiling ESP8266WiFiMulti.cpp...
[8.6%] Compiling ESP8266WiFiScan.cpp...
[10.0%] Compiling ESP8266WiFiSTA.cpp...
[11.4%] Compiling WiFiClient.cpp...
[12.9%] Compiling WiFiClientSecure.cpp...
[14.3%] Compiling WiFiServer.cpp...
[15.7%] Compiling WiFiUdp.cpp...
[17.1%] Compiling server_drv.cpp...
[18.6%] Compiling spi_drv.cpp...
[20.0%] Compiling wifi_drv.cpp...
[21.4%] Compiling WiFi.cpp...
[22.9%] Compiling WiFiClient.cpp...
[24.3%] Compiling WiFiServer.cpp...
[25.7%] Compiling WiFiUdp.cpp...
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:29:0:
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.h:21:7: error: redefinition of 'class WiFiUDP'
 class WiFiUDP:public UDP
       ^
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/utility/wifi_drv.h:22:0,
                 from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:26:
/Users/eddy/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.h:32:7: error: previous definition of 'class WiFiUDP'
 class WiFiUDP : public UDP, public SList<WiFiUDP> {
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In constructor 'WiFiUDP::WiFiUDP()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:35:22: error: class 'WiFiUDP' does not have any field named '_sock'
 WiFiUDP::WiFiUDP() : _sock(NO_SOCKET_AVAIL) {}
                      ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint8_t WiFiUDP::begin(uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:45:9: error: '_sock' was not declared in this scope
         _sock = sock;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:46:9: error: '_port' was not declared in this scope
         _port = port;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::available()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:56:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual void WiFiUDP::stop()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:66:8: error: '_sock' was not declared in this scope
    if (_sock == NO_SOCKET_AVAIL)
        ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:69:26: error: '_sock' was not declared in this scope
    ServerDrv::stopClient(_sock);
                          ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::beginPacket(IPAddress, uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:88:7: error: '_sock' was not declared in this scope
   if (_sock == NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:90:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:101:32: error: '_sock' was not declared in this scope
  return ServerDrv::sendUdpData(_sock);
                                ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual size_t WiFiUDP::write(const uint8_t*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:111:27: error: '_sock' was not declared in this scope
  ServerDrv::insertDataBuf(_sock, buffer, size);
                           ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:125:23: error: '_sock' was not declared in this scope
    ServerDrv::getData(_sock, &b);
                       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read(unsigned char*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:137:31: error: '_sock' was not declared in this scope
    if (!ServerDrv::getDataBuf(_sock, buffer, &size))
                               ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::peek()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:152:22: error: '_sock' was not declared in this scope
   ServerDrv::getData(_sock, &b, 1);
                      ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual IPAddress WiFiUDP::remoteIP()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:166:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint16_t WiFiUDP::remotePort()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:176:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
[Build] Error occurred.`

eykamp, 2017-05-30:
For what it's worth, I am seeing this as well.  It looks to me as if the compiler is trying to compile both the ESP8266 wifi libs and the standard Ardino wifi libs, and they are colliding.  I suspect the solution is to omit the Arduino libs.  I am not sure how this works under the sheets, but I do know that the same code compiles fine in the Arduino IDE.  I have the correct platform and board selected in Sublime.

cooperbaker, 2017-06-27:
I am also having this issue.  ESP8266 examples do not compile with Stino, but they compile just fine with Arduino IDE.  Looks like library files are being omitted and redeclared during compile.

Anyone know a workaround?

Robot-Will, 2017-07-15:
Sorry about this problem. You can revise the file [plugin folder]/libs/stino_runtime/__init__.py and it will work.

Change line 1046:
platform_paths = [platform_path, sketchbook_path, ext_app_path]

to:
platform_paths = [platform_path, sketchbook_path]

Thanks.

ESP8266user, 2018-08-09:
I'm also not able to use Sublime due to this error. The fix suggested by Robot-Will is not working for me.

Using sublime on Macbook with OSX 10.13.3 
I tried esp8266 library 2.40 till 2.4.2.

Build] /Users/ivojanssen/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/examples/WiFiWebServer...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling WiFiWebServer.ino.cpp...
[2.7%] Compiling WiFiClientSecure.cpp...
[4.1%] Compiling ESP8266WiFiGeneric.cpp...
[5.4%] Compiling ESP8266WiFiMulti.cpp...
[6.8%] Compiling WiFiServer.cpp...
[8.1%] Compiling ESP8266WiFiSTA.cpp...
[9.5%] Compiling WiFiClient.cpp...
[10.8%] Compiling ESP8266WiFiScan.cpp...
[12.2%] Compiling ESP8266WiFiAP.cpp...
[13.5%] Compiling ESP8266WiFi.cpp...
[14.9%] Compiling WiFiUdp.cpp...
[16.2%] Compiling WiFiServerSecure.cpp...
[17.6%] Compiling spi_drv.cpp...
[18.9%] Compiling server_drv.cpp...
[20.3%] Compiling wifi_drv.cpp...
[21.6%] Compiling WiFiServer.cpp...
[23.0%] Compiling WiFiClient.cpp...
[24.3%] Compiling WiFiUdp.cpp...
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:29:0:
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.h:27:7: error: redefinition of 'class WiFiUDP'
 class WiFiUDP : public UDP {
       ^
In file included from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/utility/wifi_drv.h:26:0,
                 from /applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:26:
/Users/ivojanssen/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiUdp.h:32:7: error: previous definition of 'class WiFiUDP'
 class WiFiUDP : public UDP, public SList<WiFiUDP> {
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In constructor 'WiFiUDP::WiFiUDP()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:35:22: error: class 'WiFiUDP' does not have any field named '_sock'
 WiFiUDP::WiFiUDP() : _sock(NO_SOCKET_AVAIL) {}
                      ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint8_t WiFiUDP::begin(uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:45:9: error: '_sock' was not declared in this scope
         _sock = sock;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:46:9: error: '_port' was not declared in this scope
         _port = port;
         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::available()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:56:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual void WiFiUDP::stop()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:66:8: error: '_sock' was not declared in this scope
    if (_sock == NO_SOCKET_AVAIL)
        ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:69:26: error: '_sock' was not declared in this scope
    ServerDrv::stopClient(_sock);
                          ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::beginPacket(IPAddress, uint16_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:88:7: error: '_sock' was not declared in this scope
   if (_sock == NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:90:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:101:32: error: '_sock' was not declared in this scope
  return ServerDrv::sendUdpData(_sock);
                                ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual size_t WiFiUDP::write(const uint8_t*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:111:27: error: '_sock' was not declared in this scope
  ServerDrv::insertDataBuf(_sock, buffer, size);
                           ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:125:23: error: '_sock' was not declared in this scope
    ServerDrv::getData(_sock, &b);
                       ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read(unsigned char*, size_t)':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:137:31: error: '_sock' was not declared in this scope
    if (!ServerDrv::getDataBuf(_sock, buffer, &size))
                               ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::peek()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:152:22: error: '_sock' was not declared in this scope
   ServerDrv::getData(_sock, &b, 1);
                      ^
[... cut ...]
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:166:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp: In member function 'virtual uint16_t WiFiUDP::remotePort()':
/applications/Arduino.app/Contents/Java/libraries/WiFi/src/WiFiUdp.cpp:176:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
[Build] Error occurred.

Diex, 2019-07-18:
you have to restart sublime in order to work. :)

Pierrickgentil, 2020-05-09:
Hi,

I have install stino in Sublime Text 3 as the " README" folder say.
I am using ESP8266 version 2.5.0 library inside of Stino. When i try to build my code there is troubles. The code works perfect with Arduino IDE.

The error is : 

[Build] C:/Users/PC/Documents/Arduino/Codes/Mes codes/Serveur_WEB...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[10.0%] Compiling server_drv.cpp...
[20.0%] Compiling spi_drv.cpp...
[30.0%] Compiling wifi_drv.cpp...
[40.0%] Compiling WiFi.cpp...
In file included from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.h:32:0,
                 from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.cpp:21:
C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFiServer.h:37:14: error: invalid abstract return type for member function 'WiFiClient WiFiServer::available(uint8_t*)'
   WiFiClient available(uint8_t* status = NULL);
              ^
In file included from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.h:31:0,
                 from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.cpp:21:
C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFiClient.h:27:7: note:   because the following virtual functions are pure within 'WiFiClient':
 class WiFiClient : public Client {
       ^
In file included from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFiClient.h:24:0,
                 from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.h:31,
                 from C:/Program Files (x86)/Arduino/libraries/WiFi/src/WiFi.cpp:21:
C:/Users/PC/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Client.h:29:21: note: 	virtual int Client::connect(const IPAddress&, uint16_t)
         virtual int connect(CONST IPAddress& ip, uint16_t port) =0;
                     ^
C:/Users/PC/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)
         virtual bool flush(unsigned int maxWaitMs = 0) = 0;
                      ^
C:/Users/PC/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)
         virtual bool stop(unsigned int maxWaitMs = 0) = 0;
                      ^
[Build] Error occurred.

Please, help me