Robot-Will/Stino #446: Sublime performance issue ?

dreammonkey, 2017-05-13:
hi there !

nice work, but when I install your package on Mac, Sublime Text 3, Sublime becomes very unresponsive, nearly unusable :/ any idea why this could be happening ?

cheers,
D

42loop, 2017-05-23:
hi,
experiencing same on ubuntu 16.04, often it becomes gray and takes 10 secs or so to close.
today i tried to use sublime without network connection and it wouldn't even start up...

grtx

SebastienDaniel, 2017-07-14:
Same on Arch Linux

Robot-Will, 2017-07-15:
No idea. Maybe I could find out what's wrong.

geonicz, 2017-09-21:
Same issue on Manjaro linux (Arch derivative)
uname -a :  Linux god-pc 4.11.12-1-MANJARO #1 SMP PREEMPT Fri Jul 21 08:51:46 UTC 2017 x86_64 GNU/Linux.

SephReed, 2017-12-23:
I've found a few interesting things in regards to this error.  Here's the info I have so far:

- On OSX the process is plugin_host
- Undo (Meta-Z) is incredibly slow after a fresh upload, as are the first few updates to the text.  After the first few (very laggy) updates to the text, everything is fine, so I always do a keyboard solo before doing undos.
- Hovering over the "Serial Monitor" option is guaranteed lag.

Overall, it's not as big of a deal as it had been at first.  The trick which works for me is to just press a bunch of keys until it gives up on whatever it thinks it should be doing in response to typing, and from then on it no longer lags on key input or file scrolling.  

I have a very strong suspicion that there is some code (outside of build) which cares about the context of the text file and it is the culprit.