Robot-Will/Stino #447: Install Steps

InvisiBug, 2017-05-17:
I have tried to install Stino but when I try to do anything the menu seems broken. Am I doing something wrong?

[image]

eshira, 2017-09-29:
I had the same issue, and I solved it eventually, so let me try to recreate my steps. I was not taking notes, just clicking around submenus while half-asleep... so I'm not completely sure if this is the solution.

Firstly, I believe I went to Arduino > Install platform > arduino > Arduino AVR boards
Secondly, I believe I went to Arduino > Platform > arduino > Arduino AVR boards and selected that option (this unchecks the other option which is Arduino IDE > Arduino)
Thirdly, I may have had to restart the software. I did restart it a few times and I don't know if that affected anything.
At some point I noticed boards were showing up in the Board menu.

Destiny9701, 2017-11-22:
How can I add a lib?

closed 2020-07-27 by InvisiBug