Robot-Will/Stino #448: ESP8266WiFiMulti build error, 'INT_MIN' was not declared...

hodavame, 2017-05-21:
First of all, I want to thank you for your great work!

The problem: I can not compile with the ESP8266 platform. In Arduino it works fine.

I tried deleting everything, Stino and platforms... then I installed Stino cloning the repository and then I installed the ESP8266 and Arduino platforms from Stino (not Arduino), and I still get exactly the same error, even with the examples included in the platform.

Error:
`[6.2%] Compiling ESP8266WiFiMulti.cpp...
C:/Users/hodav/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::run()':
C:/Users/hodav/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp:55:33: error: 'INT_MIN' was not declared in this scope
             int bestNetworkDb = INT_MIN;
                                 ^
[Build] Error occurred.`

I appreciate any clues you can give me to avoid this problem!

dombic, 2018-01-02:
Same problem in Arduino 1.8.5 IDE.
Strange, as limits.h is being included at the top of ESP8266WiFiMulti.cpp.

RoadkillUK, 2018-04-08:
Any new on this as I have the same problem now :(

sstefanov, 2018-10-24:
Same problem!