Robot-Will/Stino #455: How to config the key bindings?

segatecm, 2017-06-11:
I want to config the Stino shortcut keys like arduino IDE.
such as Ctrl + U for upload the sketch.

gepd, 2017-06-12:
Open the key bindings file (`Preferences > Key Bindings`) and paste this

```
{ "keys": ["ctrl+u"], "command": "stino_upload" },
{ "keys": ["ctrl+b"], "command": "stino_build" }
```

segatecm, 2017-06-13:
thanks @gepd, it works!

so if i want bind more function,  where can i find the functions define? such as monitor start/stop.

gepd, 2017-06-13:
You can use the menu file as reference: https://github.com/Robot-Will/Stino/blob/Stino-Dev/Main.sublime-menu.

closed 2017-06-21 by segatecm

Maik93, 2017-08-31:
Sorry, if you link ctrl+b with Stino, it will allways run instead of Sublime Build (i.e. gcc for C files)?

segatecm, 2017-09-06:
No, That not hanpped. Stino has the config file for itself.