Robot-Will/Stino #458: Keywords within comments (like Serial) are highlighted in syntax color / override the comment color.

derekmtran, 2017-07-06:
[image]

Ideally (maybe this is preference so this isn't truly an issue), words like Serial would not be highlighted. It would be nice to have it greyed out to be visually obscured when looking at _active_ the code at a glance.

nuwantha2020, 2017-07-22:
I also noted this problem...
with some modifications to syntax file it could be corrected. 

The problem is due to the key, "with_prototype" used in Arduino.sublime-syntax which makes all the list of match keys to be included in each and every context in c++.sublime-syntax file. So comments context is ignored for match keys in Arduino.sublime-syntax. Thus they appear highlighted even commented. 

This can be corrected for now by adding full C comment context to the Arduino.sublime-syntax file. Then it will also check for comments.

I'm attaching the corrected Arduino.sublime-syntax file to this. Replace original Arduino.sublime-syntax file found in package folder with this one. 
[Corrected_Arduino.sublime-syntax.zip](https://github.com/Robot-Will/Stino/files/1167373/Corrected_Arduino.sublime-syntax.zip)

Jshn245, 2017-07-29:
I've also noticed that it does not highlight the byte keyword.. others might be missing too.