Robot-Will/Stino #463: Make Sublime Text interface slow and laggy

giogziro95, 2017-07-25:
I'm not sure if it's a known issue or not. I've searched for it and I can't find one, so I'm submitting my own.

When I install Stino by dropping the folder inside _Packages_ directory, Sublime Text interface becomes slow and laggy to the point that it becomes almost unusable — opening and closing menus is slow, closing a tab is slow, closing a window is extremely slow. I noticed that when I install it, it also creates another Stino folder inside the _Packages/User_ directory, so I tried to drop the main folder inside _User_; as a result, it worked and was no longer slow. Is it just me or is it a known issue? In [README.md § Installation](https://github.com/Robot-Will/Stino#installation), it says to copy it inside _Packages_, not _Packages/User_, so I don't think I'm doing something wrong.

I'm wondering why it's slower in one case and not in the other. What's the difference between the two directories?

Jshn245, 2017-07-29:
Exact same problem for me. I think there's a memory leak somewhere.

giogziro95, 2017-07-30:
@Jshn245 In [README.md § Requirements](https://github.com/Robot-Will/Stino#requirements), it says that it requires Sublime Text 3, Build 3126, so I don't think that's the case.

mattmon, 2017-08-16:
I can confirm this in Build 3126 running on Debian 9.
Installing this plugin makes Sublime Text unusable and slow. 
Remove the plugin and problem disappears.

SMemsky, 2017-08-25:
Hey, same here. Build 3126 on ArchLinux. Removing the plugin problem disappears, but Arduino submenu still appears on the top with all it's items marked as unavailable.

giogziro95, 2017-08-25:
Forgot to mention that my OS is Manjaro Linux 17 (Gellivara). Since two more people commented that they use Linux, I think the issue could be Linux-specific.

nash169, 2017-09-08:
I have the same problem on macOS...

giogziro95, 2017-09-14:
Doesn't appear to be replicable on Windows 10.

geonicz, 2017-09-21:
Same issue on Manjaro Linux(Arch derivative)
uname -a: [Linux god-pc 4.11.12-1-MANJARO #1 SMP PREEMPT Fri Jul 21 08:51:46 UTC 2017 x86_64 GNU/Linux]
I'm wandering: if there any way to run plugin under profiler and find out where leakage happens?

maackle, 2017-10-07:
This is a bit more specific, but when do a Build and then there are lots of errors that show in the editor, ST completely freezes up, making me force quit.

kallere, 2017-10-19:
I have the same issue and symptoms as previous posters on Mac OS 10.11.6 and Sublime 3.0 build 3143

giogziro95, 2017-10-19:
I recommend switching to [Atom and PlatformIO](https://www.youtube.com/watch?v=EIkGTwLOD7o). That's what I did and I like it a lot.

kallere, 2017-10-19:
@giogziro95 Thank you for great tip - was not aware of this before and initial impression is very good.

shpiki, 2017-10-30:
same here, Sublime becomes unusable completely and needs to force quit, OS is Linux Mint 18.2

thundR, 2017-11-14:
Is this issue being worked on? Have the same problem on arch.

geonicz, 2017-11-21:
Same lags as before on arch linux. Just cheched it again(installed from ST repo using package manager). Thought about profiling stino code & getting rid of lags but got Deviot. A bit worse exactly for arduino, but ok and works fine. Just because ST gets unusable at all with Stino. So, maybe, it's better for me to migrate. Anyway, thanks to Robot-Will for a very nice plugin!

ogermer, 2017-12-04:
I can confirm the lag issue with Sublime Text 3.0 Build 3143.
Sometimes it works - and suddenly I get beachballed on my Mac.
Is Deviot a better platform? I tend to like the Stino UI better. Any thoughts on this?
And BTW: Is Stino still under active development? The number of people in this issue makes me wonder...?

giogziro95, 2017-12-04:
@geonicz Thanks for mentioning Deviot! I've just installed it and it's basically a lightweight version of PlatformIO. It might come in handy on low-performance machines.

rgbink, 2017-12-11:
Landed here looking for solution, but will add my +1 to the issues already reported: CPU spike and unusable UI until I have to force quit.

curtiszimmerman, 2018-01-30:
Build 3143, LM 18 Cinnamon (64-bit). ST3 is my daily driver for programming. Installed Stino package and it murdered the UI menu elements with enormous delays. It seems to be doing something like probing the Arduino every time I try to interact with an ST3 menu element. Uninstalled and moved on to try Deviot as suggested.

giogziro95, 2018-01-30:
> Latest commit a94831c  on Apr 17, 2017

This project doesn't seem to be under an active development.

AlexandrosFerles, 2019-12-10:
Anyone find  a fix on this? This problem is still active on Ubuntu 19.04

mrcarriere, 2020-12-16:
Just wanted to comment that I also ran into this as well. I have a hunch that this has something to do with the serial ports. Whenever I try to view that menu, the UI would always lag (presumably while refreshing the devices) and at times when sublime would lag, I would hear my headset announcing that it was disconnecting/reconnecting to bluetooth.

This is getting to the point where I would be disabling/uninstalling this package unless I'm actively working on an Arduino project.