Robot-Will/Stino #465: macOS not working

Diex, 2017-07-27:
on Build...

[Build] /Users/blabla/Documents/Arduino/blabla/blabla...
**[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.**

macOS Sierra 10.12
Arduino IDE Path: /Applications/Arduino.app/Contents/Java

machine:Java user$ pwd
/Applications/Arduino.app/Contents/Java
machine:Java user$ ./arduino -version
Arduino Builder 1.3.21
Copyright (C) 2015 Arduino LLC and contributors
See https://www.arduino.cc/ and https://github.com/arduino/arduino-builder/graphs/contributors
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
roma:Java diex$

Tpixelminer, 2017-11-15:
Im getting the same error on macOS Sierra :/

ogermer, 2017-12-04:
Happened to me too when I first installed - but somehow went away after Stino decided to download additional assets (saw that in the status window). Not sure if / where I triggered that - but I could use Stino until Sublime Text became unresponsive. Hate that beachball.

duregard, 2018-03-04:
This problem persists however only on Adafruit-platform (any version, osx : 10.13 ), anyone have any idea how to force asset download, or manually install a platform?

HapCoderWei, 2018-08-14:
I have just fixed this issue on my mac. And below is what i did. Hope this can give some help.
Click the Arduino -> Install Platform -> arduino -> arduino AVR Boards, and select a version of Arduino you like, then the sublimt text will download toolchain automaticlly. After all this, my problem is solved!
If it seemd doesn't download anything, just do it again or choose another version to download. I was chose 1.6.17 first, but it didn't work. Then i chose the 1.6.16, and it worked!
Hope this will help you.
[image]

h-jia, 2021-10-29:
1. delete /Users/hong/Library/Arduino15
2. Find your EXACT Arduino board info
3. Arduino->Install Platform -> Arduino -> Your EXACT Arduino and version
4. Select Board->Your EXACT board

Works on MacOS 12