Robot-Will/Stino #475: Remove Package leave Arduino menu behind

papadeltasierra, 2017-09-29:
If I use Package Control to remove this package, the Arduino menu is left behind.

eshira, 2017-09-29:
I removed Stino by manually removing these two folders: sublime-text-3 > Packages > Stino-Stino-Dev as well as sublime-text-3 > Packages > User > Stino, then restarting Sublime Text 3.

papadeltasierra, 2017-10-02:
Thanks.  At some point I have to have another go at installing Stino anyway.  It didn't go smoothly the first time and I was in a rush so simply gave up and went back to the Aduino IDE.  Would be nice to get it working though as the IDE is not exactly fully featured :-).

gepd, 2017-10-02:
You can check [this](https://github.com/gepd/Deviot) I've been working on it for a while

techydude, 2017-10-17:
Is Stino dead (again)?  I'm always enthusiastic of Will's efforts (if I were a s/w dev I'd help, but I'm not), but the last flurry of dev activity was march/april, there's 46 issues and 3 PRs since then.  Isn't this the definition of an abandoned project?  Are any of the many many forks of Stino progressing instead?