Robot-Will/Stino #479: [Error] Toolchain is not ready - after update

BER7777, 2017-10-16:
I don't know if it is related but after the most recent update of Sublime Text I can't Build or Upload to the board. It used to work like a charm. 

Here is the error that I am getting:
```
[Build] C:/Users/.../Documents/Arduino/files/Blink...
[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.
```
Here is the output from the console:
```
DPI scale: 1
startup, version: 3143 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/.../AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/.../AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/.../AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.401802
startup time: 0.500802
first paint time: 0.533802
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Colorsublime.colorsublime-plugin
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin PackageResourceViewer.package_resource_viewer
reloading plugin PackageResourceViewer.package_resources
reloading plugin Stino-Stino-Dev.StinoCommands
plugins loaded
http://downloads.arduino.cc/packages/package_index.json <urlopen error [Errno 11001] getaddrinfo failed>
http://downloads.arduino.cc/libraries/library_index.json <urlopen error [Errno 11001] getaddrinfo failed>
Package Control: Skipping automatic upgrade, last run at 2017-10-16 09:28:16, next run at 2017-10-16 10:28:16 or after
```
Any help is apreshiated.

murych, 2017-11-03:
getting this error too
- windows 8.1 professional 64 bit
- sublime text 3.0 build 3143

Tpixelminer, 2017-11-15:
Im getting the same error too on MacOS Sierra. Maybe we just need to wait for it to install lol?

pablobacho, 2017-12-04:
Same. Ubuntu 17.10 x64. Sublime build 3143. :(
```
[Build] /home/...
[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.
```
I am not sure what changed on my setup, though. Does anybody have a solution? Thanks.

Diex, 2017-12-07:
same here...
macOS 10.12
sublime 3

ckellerman, 2017-12-14:
Me as well:
- macOS Sierra 10.12.6
- Sublime 3.0, Build 3143
- Platform Adafruit Bluefruit nRF52

palmarytech, 2017-12-24:
Me same 

- win 10

- Sublime 3143

madacol, 2018-02-16:
Same here, installing for the first time.
Ubuntu 16.04
Sublime Text Build 3143

No one knows how to fix?

Momshad-Dinury, 2018-03-01:
I was having the same problem but when I was about to comment here I noticed Sublime Text was automatically downloading some files and after it finished downloading I tried uploading again and everything is working like a breeze.

Flowpy, 2018-04-07:
I just got this problem too and here is a workaround I found :
- Delete the "Arduino15" folder (Windows path : "%appdata%\..\Local\Arduino15")
- Start Sublime, then in the Arduino menu : "Install Platform" > "Refresh" (this will recreate the "Arduino15" folder)
- Then : "Install Platform" > "arduino" >"Arduino AVR Boards" > pick latest version
- Wait for download/installation and compilation will work again !

xtremeroboz, 2019-01-11:
Solution:
Download package_index.json from this link http://downloads.arduino.cc/packages/package_index.json
then put it "%appdata%..\Local\Arduino15"

sunupradana, 2019-02-05:
Still not working.

adersh14, 2019-06-13:
Thanks Flowpy..it works

adersh14, 2019-06-13:
> I just got this problem too and here is a workaround I found :
>
> * Delete the "Arduino15" folder (Windows path : "%appdata%..\Local\Arduino15")
> * Start Sublime, then in the Arduino menu : "Install [...]

It works