Robot-Will/Stino #481: Espressif esp32 board error

MyrmidonGitHub, 2017-11-10:
Hi!
**I'm new here**, trying to set your **great** Stino under Sublime Text v3.0, Build 3143, and I want to report an error when selecting **sketchbook -> espressif esp32 board**. The error is:
"Error trying to parse file: Unexpected character, expected a comma or closing bracket in Packages\User\Stino\menu\board\Main.sublime-menu:86:20". It pops-up many consecutive times (15 or 20 aprox.). After closing all error windows, I can begin editing.

The refered file content at the 86 line is:

	(line 86)"caption": ""WeMos" WiFi&Bluetooth Battery",
	(line 87)"id": "stino_board_"WeMos" WiFi&Bluetooth Battery",
	(line 88)"command": "stino_select_board",
	(line 89)"args": {"board_name": ""WeMos" WiFi&Bluetooth Battery"},
	(line 90)"checkbox": true

I attach my (compressed) full Main.sublime-menu, maybe it will help.

[Main.zip](https://github.com/Robot-Will/Stino/files/1462427/Main.zip)

I use Windows 7 x64 ultimate SP1 and I must say I installed Stino uncompressing your .zip file to "C:\Program Files\Sublime Text\Data\Packages" and I don't know if it would be better to use Package Control to install Stino.

Later I cannot make it build the basic blink program: 
(Console)
[Build] C:/Program Files (x86)/Arduino/examples/01.Basics/Blink...
[Step 1] Check Toolchain.
[Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.

but I assume it's because the first error.

**Thank you (very much) in advance**

miguelpom, 2018-01-15:
Hi @MyrmidonGitHub ,

I have just the same problem in my ST3+Stino. I installed Stino from Package Control. 
I think the error here might be in the double quotes in:
```		
{
	"caption": ""WeMos" WiFi&Bluetooth Battery",
	"id": "stino_board_"WeMos" WiFi&Bluetooth Battery",
	"command": "stino_select_board",
	"args": {"board_name": ""WeMos" WiFi&Bluetooth Battery"},
	"checkbox": true
},
```
I mean the ones surrounding the word "WeMos" throughout the file.

I tried to delete these 'extra' quotes but next time I launched Sublime Text the error persisted, as if the file "Main.sublime-menu" were regenerated with each launch of ST or the Stino package.

In my Windows10 system, I have also installed the Arduino IDE 1.8.5 (with the support packages for the [ESP8266](https://github.com/esp8266/Arduino) and [ESP32](https://github.com/espressif/arduino-esp32) boards) in parallel with ST and the Stino package.

I hope my post helps to draw some attention over this issue.
Btw, @MyrmidonGitHub , if you've made any progress solving this I would really appreciate any help.

Thank you in advance to all githubbers.

smartinick, 2018-01-29:
Same Problem here.... :(

Also the settings are not saved - i can change the app-dir.stino-settings and save it, once i restart sublime i have the default contents again.

and (i guess that's a problem caused by wrong directory settings mentioned above) custom libraries (in users documents/arduino/libraries) are ignored.

so as happy as i was to find stuino, as sad it is half an hour later as it's not usable for my case :(

nehu92, 2018-04-06:
{
	"caption": ""WeMos" WiFi&Bluetooth Battery",
	"id": "stino_board_"WeMos" WiFi&Bluetooth Battery",
	"command": "stino_select_board",
	"args": {"board_name": ""WeMos" WiFi&Bluetooth Battery"},
	"checkbox": true
},

same problem with double coutes, re generate if i deleted by hand. no solution yet?

miguelpom, 2018-05-10:
Thanks, @marciofcalil . Your solution worked perfectly for me.

nehu92, 2018-05-10:
{
"caption": ""WeMos" WiFi&Bluetooth Battery",
"id": "stino_board_"WeMos" WiFi&Bluetooth Battery",
"command": "stino_select_board",
"args": {"board_name": ""WeMos" WiFi&Bluetooth Battery"},
"checkbox": true
},

just write WeMos with out the quotation marks and that word for me

{
"caption": "WeMo" WiFi&Bluetooth Battery",
"id": "stino_board_WeMos WiFi&Bluetooth Battery",
"command": "stino_select_board",
"args": {"board_name": "WeMos WiFi&Bluetooth Battery"},
"checkbox": true
},

codework1010, 2018-06-27:
I had this issue and solved it like this. Please note that this will fix this error, but a more general solution is easily possible. 

How Error Happens:
First, let me show you how the error occurs. The Arduino system of folders expects a `hardware` folder. All programs related to an espressif device are placed here such as this `hardware\espressif\esp32`.  In the esp32 subfolder, you will find the `boards.txt` file. The `boards.txt` file contains a list of boards that use ESP32 chip. The issue occurs with the WeMos board description found inside the `boards.txt` file.

For the WeMos board, the name has a string containing double quotes. These double quotes around WeMos are cause of error for me. If you open the `board.txt` file and search for `WeMosBat.name="WeMos" WiFi&Bluetooth Battery` you can observe the double quotes.

If you open in menu data file for sublime-text (`Main.sublime-menu`) and search WeMos will observe the double quotes error persists. The file is found in the folder path `C:\sublime\...\Data\Packages\User\Stino\menu\board\Main.sublime-menu`. 

NOTE: Sublime-text may not be located in `C:\sublime\`. I installed it on my operating system in the manner. You may have it elsewhere.

How I Fix Double Quotes Error
We will ass one line of code to the Stino script that makes the sublime-text menu data. This line of code will remove any unnecessary double quotes from the board name.

Ready.. Okay, let's do this!

Stino has a python script, called `st_menu.py`.  This python file creates the sublime menu for the esp32 board. This python script is contained in a folder named `stino_runtime`. Here is the folder path to the file `C:\sublime\...\Data\Packages\Arduino-like IDE\libs\stino_runtime`.  In sublime-text, open the `st_menu.py` file. Search for the `def update_board_menu(arduino_info)` function (it located about line 974). Inside the function search for the `board_name` FOR-LOOP (see sample code below). Inject this one line of code that will strip double-quotes from any board_name into the first line of the FOR-LOOP'S body (see sample code below).

SAMPLE CODE
`
    for board_name in board_names:
        board_name = board_name.replace('"','')    # Fix: remove double quotes from board name
`
After this fix save the file. Now, close and open sublime. 

Summary of What I Learned
The `update_board_menu(arduino_info)` function is involved in the creation of the sublime board menu file found here `C:\sublime\...\Data\Packages\User\Stino\menu\board\Main.sublime-menu`. 

If open this file BEFORE fix you will see the double quotes error. The `update_board_menu` function always generate this `\board\Main.sublime-menu` file on the initialization of sublime-text. That is why any direct changes to this file will be changed again by the `update_board_menu` function. The fix must be in the  `st_menu.py` code.

Future works:
A better fix is to create a script that verifies the text provided by hardware developers is defined correctly. Here a regular expression can be used to define allowed strings. This is the more general solution than my fix. My fix removes double quotes but does not verify.

If this fixed your error, I'm happy you are going to have a great day! Now, go blink an LED in my name.

tuan-karma, 2019-03-09:
> I had the same error, so I was looking for what it could be and I'm pretty sure I’ve found the error.
> 
> To solve the problem of double quotes go to the folder:
> **/Arduino/Hardware/expressif/esp32/**
> 
> There is a file called **boards.txt.**
> 
> Open this file and search for: **WeMos WiFi&Bluetooth Battery** , in my case it was on line 381.
> 
> Remove the quotes from this line, save the file and open Sublime Text again.
> 
> **The first time I've opened Sublime, I got the same error, so if this happens, close Sublime again and return to open.**
> ***It was some cache trash (popups).
> 
> Let me know if this works for you.
> 
> Cheers.

Thank you for your solution, marciofcalil !

I remember, your solution used to work last year. But today I encounter the problem with the "Wemos" board quoting again. 

I would like to update your answer by adding the following information: the boards.txt file now should locate in the new path in windows machine. Something likes this: 
C:\Users\***YOUR_NAME***\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1

mumairanwar92, 2019-03-21:
{
    "caption": ""WeMos" WiFi&Bluetooth Battery",
    "id": "stino_board_"WeMos" WiFi&Bluetooth Battery",
    "command": "stino_select_board",
    "args": {"board_name": ""WeMos" WiFi&Bluetooth Battery"},
    "checkbox": true
},

replacing the " to \" worked for me.

{
    "caption": "\"WeMos\" WiFi&Bluetooth Battery",
    "id": "stino_board_\"WeMos\" WiFi&Bluetooth Battery",
    "command": "stino_select_board",
    "args": {"board_name": "\"WeMos\" WiFi&Bluetooth Battery"},
    "checkbox": true
},

yesnoj, 2024-05-28:
sorry for reopening this ticket, but i've still an issue , here the following error of double quotes :

						"caption": "Adafruit MagTag 2.9"",
						"id": "stino_board_Adafruit MagTag 2.9"",
						"command": "stino_select_board",
						"args": {"board_name": "Adafruit MagTag 2.9""},
						"checkbox": true
						
How can be solved!?

yesnoj, 2024-05-28:
@codework1010  thanks you!I've solved that issue with your mod!Thanks again! :)