Robot-Will/Stino #483: Uses 98% of CPU on OSX 10.11 with Sublime Text 3

SephReed, 2017-12-13:
It works pretty well, except for the 98% CPU usage.  Lags on every single keystroke, to the point of being unusable.

frankzimper, 2017-12-14:
Same here. Using ST3 on Linux (Ubuntu 16.04). When launching sublime text with the Arduino IDE enabled, the hud-service process consumes a lot of CPU power.

@SephReed can you find out which process on your Mac is causing the high CPU usage?

frankzimper, 2017-12-14:
Possible duplicate of #446 ?

yocarolo, 2018-07-04:
Every time that you upload a program from Sublime with Stino to your Arduino on OSX (I have 10.13.5) you launch a process called "avrdude", you can observe that with your "Activity Monitor" on your mac. After the upload is finish the process "avrdude" is killed and everything is OK.

NOW, If you don't have the right Serial Port selected on your Stino Software and you upload a program to an inexistent port on your Arduino, evidently is not going to work... AND this "avrdude" process is going to stay started and NEVER killed, even if you quit Sublime. 

If you repeat this sequence three times you are going to create there different "avrdude" processes that are never going to be killed. They consume all your battery and CPU and they super heat the computer, the fan starts to go crazy. 

Even if you quit "Sublime" the processes are going to stay there until you kill them or restart the computer. If you change to the right Serial Port and upload again, even if this time is working the old "
[image]
avrdude" process are going to stay alive.

The only solution that I found was to open your "Monitor" software and kill them manually or restart your computer. 

I hope this can be useful for you guys and I hope that the developers will fix this soon.

[image]