Robot-Will/Stino #484: Cannot compile code that contains ISR() macro

Warwolt, 2017-12-26:
I tried writing a sketch that contained 
```
ISR(TIMER1_COMPA_vect)
{
    digitalWrite(13, HIGH);
}
```
When I compile this using the Arduino IDE, it works fine and I can flash and run the code. If I try to compile this inside Sublime 3 using Stino, I get an error like this:

> 
[Build] C:/Users/rasmu/Documents/Arduino/led_pong...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[50.0%] Compiling led_pong.ino.cpp...
In file included from C:/.../arduino/Arduino.h:30:0,
                 from C:/.../led_pong/led_pong.ino:15:
C:/.../led_pong/led_pong.ino:58:1: error: conflicting declaration of 'void __vector_11()' with 'C' linkage
 ISR(TIMER1_COMPA_vect)
 ^
C:/.../led_pong/led_pong.ino:19:6: note: previous declaration with 'C++' linkage
 void setup()
> 

Something seems to be happening with the preprocessing of the ISR() macro?

SterlingPeet, 2018-02-21:
I can confirm that I have this exact same behavior on my OSX.  I have been able to work around it by placing `extern "C" {` and `}` around my ISRs.  Its pretty obnoxious, but it seems to compile in both Stino and the regular Arduino IDE.

miaoudak, 2018-09-22:
I am facing the same issue. The following example compiles on arduino IDE but with stino I get the error "error: conflicting declaration of 'void __vector_11()' with 'C' linkage"
any help??

```
boolean blink;
void setup()
{
  // put your setup code here, to run once:
  Serial.begin(9600);
  initMyTimer();
}

void loop()
{
  // put your main code here, to run repeatedly:
  delay(1000);
}

void initMyTimer(void)
{
  noInterrupts(); // disable all interrupts
  TCCR1A = 0;
  TCCR1B = 0;
  TCNT1 = 0;
  OCR1A = 250; // compare match register 16MHz/64/250 results in 1ms/Tick
  TCCR1B |= (1 << WGM12); // CTC mode
  TCCR1B |= (1 << CS11); // 64 prescaler
  TCCR1B |= (1 << CS10);
  TIMSK1 |= (1 << OCIE1A); // enable timer compare interrupt
  // timer 2 for custom PWM
  interrupts(); // enable all interrupts
}

ISR(TIMER1_COMPA_vect)
{
  // timer compare interrupt service routine
  static unsigned int OneSecTimer = 1000;
  if (OneSecTimer > 0)
  {
    OneSecTimer--;
  }
  else
  {
    OneSecTimer = 1000;
    blink = !blink;
    Serial.println("tick");
  }
}
```

miaoudak, 2018-09-22:
Finally I managed to compile it using the code (separate declaration of ISR...):

```
boolean blink;
void setup()
{
  // put your setup code here, to run once:
  Serial.begin(9600);
  initMyTimer();
}

void loop()
{
  // put your main code here, to run repeatedly:
  delay(1000);
}

void initMyTimer(void)
{
  noInterrupts(); // disable all interrupts
  TCCR1A = 0;
  TCCR1B = 0;
  TCNT1 = 0;
  OCR1A = 250; // compare match register 16MHz/64/250 results in 1ms/Tick
  TCCR1B |= (1 << WGM12); // CTC mode
  TCCR1B |= (1 << CS11); // 64 prescaler
  TCCR1B |= (1 << CS10);
  TIMSK1 |= (1 << OCIE1A); // enable timer compare interrupt
  // timer 2 for custom PWM
  interrupts(); // enable all interrupts
}
extern "C" ISR(ISR(TIMER1_COMPA_vect));

ISR(TIMER1_COMPA_vect)
{
  // timer compare interrupt service routine
  static unsigned int OneSecTimer = 1000;
  if (OneSecTimer > 0)
  {
    OneSecTimer--;
  }
  else
  {
    OneSecTimer = 1000;
    blink = !blink;
    Serial.println("tick");
  }
}
```

emremutlu44, 2019-03-18:
> I can confirm that I have this exact same behavior on OSX. I have been able to work around it by placing `extern "C" {` and `}` around my ISRs. Its pretty obnoxious, but it seems to compile in both Stino and the regular Arduino IDE.

This has saved me. I want to share my simple code in here which make sublime have a hard time compiling by giving the error: conflicting declaration of 'void __vector_11()' with 'C' linkage  ISR(TIMER1_COMPA_vect)

`int timerCounter1;
int timerCounter2;
void setup()
{
	pinMode(13, OUTPUT);
	initMyTimer();
}

void loop()
{
}

void initMyTimer(void)
{
	// ------------------------------------------------------------------------------------------------------
	// TIMER INTERRUPT SETUP
	// ------------------------------------------------------------------------------------------------------
	// initialize Timer1
	cli(); // disable global interrupts
	TCCR1A = 0; // set entire TCCR1A register to 0
	TCCR1B = 0; // same for TCCR1B
	// set compare match register to desired timer count: every 0.100032 ms
	OCR1A = 1562;
	// turn on CTC mode:
	TCCR1B |= (1 << WGM12);
	// Set CS10 and CS12 bits for 1024 prescaler:
	TCCR1B |= (1 << CS10);
	TCCR1B |= (1 << CS12);
	// enable timer compare interrupt:
	TIMSK1 |= (1 << OCIE1A);
	sei(); // enable global interrupts
}

extern "C"
{
	ISR(TIMER1_COMPA_vect)
	{
		timerCounter1++;
		timerCounter2++;
		if (timerCounter1 == 30)
		// LIGHT ON @ 3000
		{
			digitalWrite(13, HIGH);
		}
		else
			if (timerCounter1 ==40)
			// LIGHT OFF @ 3400
			{
				digitalWrite(13, LOW);
				timerCounter1 = 0;
			}
	}
}
`