Robot-Will/Stino #488: sublime text arduino module mysensor code

mica12, 2018-01-18:
hi guys,

i want to change my ide and use sublime text 3.
i have installed stino and the building code is working fine with arduino standart example code like flash led but not with mysensor code during the building i can read :
[Build] C:/Program Files (x86)/Arduino/examples/MySensors/examples/RepeaterNode...
[Step 1] Check Toolchain.
[Step 2] Find all source files.

is someone is using sublime text 3 to create mysensor arduino object?

thanks a lot

michael

gepd, 2018-02-17:
You can try [Deviot](https://github.com/gepd/Deviot) 

I don't know when `Robot-Will` will back to the development of the plugin again. If you have any problem with Deviot I can help you with that.

mica12, 2018-02-25:
hi,

thanks a lot i will try

Schroedi71, 2018-08-26:
Hello ,

i have the same problem.
Is there a solution? 
Deviot is also not so good for me.
The problem exist only when using the Mysensor libary...

Frank

mica12, 2018-08-26:
hi,

i write my code with sublime text and i use arduino ide to upload the code
i have tried a lot of things but i dismissed :-( sorry

good day

michael

----- Mail original -----