Robot-Will/Stino #490: app_dir.stuino-settings not saved....

smartinick, 2018-01-29:
everytime i restart sublime i'm back with the old/default settings file, so it's not really usable :(

any idea how to track the error down to the root...? as there's no other people with this issue it seems it's caused by my setup...

Thanks for any time/hints how to resolve that!

frankzimper, 2018-01-29:
Hard to say with the information given. Maybe a permission problem?

smartinick, 2018-01-30:
Yes, but i have no idea how to get more information ;)
Running as Admin does not solve the problem.

is there anyway to get some debug infos/insight in what's happening e.g. when save the settings file...?

as no error message is shown to me, i have no idea how to provide more infos.... and no handles for further research about possible solutions :(

smartinick, 2018-01-31:
some more testing done.... fresh installation of sublime, no other plugins - same situation. different machine fresh installation - same situation. running with admin rights - no change, problem persists.
watching the console i found no errors printed.
as i'm running on windows but saw paths like "/C/users/"... instead of "C:\users\..." i gave it a try and put the folders in the settings file using the /C/... style.
that makes a difference as the settings-file is not reverted to default automatically, but looking into the boards submenu there's no boards listed (which are when i'm using the c:\... path style).

so from my point i have no idea what to test/how to debug anything... and re-configuring stino on every fresh sublime start renders it unusable for me.

any ideas how to debug what's going on there?

gepd, 2018-02-17:
You can try [Deviot](https://github.com/gepd/Deviot) 
I don't know when `Robot-Will` will back to the development of the plugin again

smartinick, 2018-02-18:
Thanks for the hint to Deviot - after some install hiccups regarding python version it's amazing. Only one issue as OTA/Devices is not showing all devices and no hostnames. Will research on that in some days.... THANKS!

closed 2018-02-18 by smartinick

gepd, 2018-02-18:
Thanks for the info, I'll check it

smartinick, 2018-02-18:
with not all devices showing it might also be my raspberry pi hostapd screwing up again. so please do not waste your time on this. if i can definetly say it's not on my end, i'll open an issue for Deviot. But the part where it lists "Generic (<ip>)" or "NodeMCU (<IP>)" would be very nice when it uses the hostname (not the ip/dns hostname, but the one set by ArduinoOTA.setHostName) instead :D Thanks!