Robot-Will/Stino #491: The Arduino Menu in MenuBar remains even after removal of the plugin

AkshaY2039, 2018-02-02:
I recently loaded the plugin to my Sublime Editor but since it became slow, I removed the package using Sublime Package Control. But the Menu Still Persists. How to get rid of that?

AngelOrea, 2018-03-26:
On macOS go to: ~/Library/Application Support/Sublime Text 3/Package/User and delete the Stino Folder. 
On windows, referring to ST3 documentation ,should be near  the next path: %APPDATA%\Sublime Text 3

AkshaY2039, 2019-05-29:
@AngelOrea Anything that can be done for Linux (Linux Mint)

cherbon1, 2019-06-03:
On Linux you'll find the directory in: ~/.config/sublime-text-3/Packages/User/Stino

Removing that directory fixed it for me.

Piotr1103, 2022-05-29:
I got the same problem, too. But on Win10...
Where can I remove that?