Robot-Will/Stino #499: [Heads up] Issue with function declaration at the top of the program

CptAwe, 2018-06-15:
In the arduino IDE if you write a function after the setup() and loop() functions then you also have to declare it at the top. For example:

Improper way:
```c
void setup(){
    // random code
}

void loop(){
    // more random code
    func1();
}

void func1(){
    // even more random code
}
```

Proper way:
```c

// function declaration
void func1();

void setup(){
    // random code
}

void loop(){
    // more random code
    func1();
}

void func1(){
    // even more random code
}
```

As of now I have never had a problem regarding this and everything was running smoothly. When I used the same code on the arduino IDE errors appeared. I have forgotten this quirk and got used to the "wrong" way of declaring a function. I believe that this way of function declaration should be discouraged.

I understand that this issue is of low importance and doesn't in the slightest diminish the great work that was put into this plugin. I just wanted to make sure that it is reported.

mp3tobi, 2018-06-21:
Well. You could also get rid of it, if you just write all your functions ABOVE setup() and loop().
This way the compiler always knows all functions since it has read all the functions before it enters the setup() or loop().

CptAwe, 2018-06-21:
You are right, of course! In my humble opinion, it is best for the code to be compatible with other compilers and relevant to the official documentation.

mp3tobi, 2018-06-21:
Ok, but since this plugin hasn't seen any new commit since one year I thought this could be a workaround for you!
Arduino is nice, but has some strange quirks in it. That's the real reason why every experienced programmer like you and me want a better editor. :)

CptAwe, 2018-06-21:
Oh stop it! I'm blushing!!!

*Please, continue*

Anyway, really, one year? That's interesting... I will put it in my to do list to take a look at this plugin in the (far) future.