Robot-Will/Stino #513: error when build esp8266 webserver

huyhtc, 2018-11-02:
hi
Error when i build esp8266 example HelloServer

[23.9%] Compiling ESP8266mDNS.cpp...
C:/Users/huydl/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
<command-line>:0:15: error: 'ESP8266_GENERIC' was not declared in this scope
C:/Users/huydl/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp:973:44: note: in expansion of macro 'ARDUINO_BOARD'
   addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
                                            ^
[Build] Error occurred.

i change __init__.py line 1046 to
platform_paths = [platform_path, sketchbook_path]
it does not work.

xabileiro, 2018-11-25:
I have the same problem whit that library

kindzhon, 2020-10-22:
no answer?

kindzhon, 2020-10-22:
I found way to be ok.           It's 2.4.2 bug.
u can edit ESP8266mDNS.cpp  like this.
/**
 * STRINGIZE
 */
#ifndef STRINGIZE
    #define STRINGIZE(x) #x
#endif
#ifndef STRINGIZE_VALUE_OF
    #define STRINGIZE_VALUE_OF(x) STRINGIZE(x)
#endif

  addServiceTxt("arduino", "tcp", "board", STRINGIZE_VALUE_OF(ARDUINO_BOARD));