Robot-Will/Stino #514: Sublime text not responding

Deepanys, 2018-11-11:
After adding stino to sublime . It became very slow and when i click on exit , not responding - force stop pop up is displayed

Deepanys, 2018-11-11:
I am using kali Linux

silbermantel, 2021-01-26:
this can be reproduced on fresh installations of ubuntu.
after launch sublime text will be unresponsive for 20-40 seconds (depending on the machine, after that it is laggy and freezes unpredicatbly

WolfElkan, 2021-12-12:
I've found this is usually triggered by using the arrow keys the first time after opening Sublime or uploading code to the Arduino.  A temporary workaround is to simply not use the arrow keys, and navigate only with the mouse.  However, this is inconvenient and easy to forget, some Sublime features (e.g. parallel editing) only work with the arrow keys, and there ought to be a better way to solve this.

I believe during this "frozen" period, Stino is reuploading the code to the Arduino, since it restarts execution once the period ends.  Enabling USART prevents the Arduino from restarting like this, but does not reduce the frozen period.  The frozen period persists even when the device is not selected, or no device is plugged in, and even when editing files unrelated to the Arduino.

A fix for this would be greatly appreciated.  Sublime is just generally slower since installing Stino.  I used to use it as an all-purpose lightweight plain text editor, but it's no longer fast enough.  Is it because of all the extra files for every version of Arduino?  If so, perhaps it could be solved by letting users download only the files for the version they need.

I'm using macOS Mojave 10.14.6

alejandro-alzate, 2024-08-24:
Same issue on linux mint as you can see it is a cpu bottleneck:
[image]
25% is 100% of 1 out of 4 cores since sublime looks to be single thread:
[image]

alejandro-alzate, 2024-08-24:
Also i just restarted sublime and this fella popped up 
[image]
Besides the fact this popped from nowhere the cpu load dropped when i clicked OK
And dear god error on line seventeen thousand one hundred thirty..