Robot-Will/Stino #516: Analog Pins (A0-A5) can not read in .h files

octarudin, 2018-12-29:
Hello, i am octa. first, i want to say thanks to robot-will and his project (stino). i feel more comfortable when i write an arduino code. you did a great job, man! okay, go to the topic. 
i have a little problem with stino. i write a library with name "variables.h". All variables is written in there included analog pins (A0 to A5). when i build the code, stino find an error. it writes **"error: 'A4' was not declared in this scope"**
but
if i write all of my variables in one file, in one big .ino file, stino works properly. 
what can i do to solve this problem ? thanks for advance

Flowpy, 2019-03-03:
Old post but for anyone reading:
I believe it's only because `#include <Arduino.h>` is missing in your .h file.

octarudin, 2019-03-03:
> Old post but for anyone reading:
> I believe it's only because `#include <Arduino.h>` is missing in your .h file.

problem solved. thanks @Flowpy

generalpf, 2019-10-26:
@octarudin you can close this issue then.

closed 2019-10-26 by octarudin