Robot-Will/Stino #519: install platform esp8266  and build with esp32 platform bug

ttlg59, 2019-03-27:
Hi everyone, i have some issue with esp8266 and esp32 platform. 

With esp8266, when i add both esp8266 and sparkfun package link to stino package list.
ST3 only find esp8266 platform in sparkfun package in both 2 line is "**SparkFun ESP8266 Boards --> 2.1.0**". Like image below: [
[image]
](url)

The second my issue is build with esp32 platform. ST noticed that **'{tools.gen_esp32part.cmd}' is not recognized as an internal or external command,**

[image]

So, please help me solve those issues.

Crapworks, 2019-06-05:
I have the same problem: "'{tools.gen_esp32part.cmd}' is not recognized as an internal or external command,"

Macklysterstofel, 2019-10-18:
Hello! I just had the same problem and I did not find any solution for that. 
Please, anyone that have some answers, I am keep looking here.