Robot-Will/Stino #523: '{tools.gen_esp32part.cmd}' is not recognized as an internal or external command, operable program or batch file.

TheRookieNerd, 2019-12-03:
I'm totally new here,

I followed the instructions while installing Stino. 
I got the following when I ran [this code](https://pastebin.com/aLZEzpBJ)

`
[Build] C:/Users/raina/Desktop/ESP32/R0...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[2.0%] Compiling R0.ino.cpp...
[4.0%] Compiling ETH.cpp...
[6.0%] Compiling WiFi.cpp...
[8.0%] Compiling WiFiAP.cpp...
[10.0%] Compiling WiFiClient.cpp...
[12.0%] Compiling WiFiGeneric.cpp...
[14.0%] Compiling WiFiMulti.cpp...
[16.0%] Compiling WiFiScan.cpp...
[18.0%] Compiling WiFiServer.cpp...
[20.0%] Compiling WiFiSTA.cpp...
[22.0%] Compiling WiFiUdp.cpp...
[24.0%] Compiling PubSubClient.cpp...
[26.0%] Compiling base64.cpp...
[28.0%] Compiling cbuf.cpp...
[30.0%] Compiling Esp.cpp...
[32.0%] Compiling FunctionalInterrupt.cpp...
[34.0%] Compiling HardwareSerial.cpp...
[36.0%] Compiling IPAddress.cpp...
[38.0%] Compiling IPv6Address.cpp...
[40.0%] Compiling main.cpp...
[42.0%] Compiling MD5Builder.cpp...
[44.0%] Compiling Print.cpp...
[46.0%] Compiling Stream.cpp...
[48.0%] Compiling StreamString.cpp...
[50.0%] Compiling WMath.cpp...
[52.0%] Compiling WString.cpp...
[54.0%] Compiling cdecode.c...
[56.0%] Compiling cencode.c...
[58.0%] Compiling esp32-hal-adc.c...
[60.0%] Compiling esp32-hal-bt.c...
[62.0%] Compiling esp32-hal-cpu.c...
[64.0%] Compiling esp32-hal-dac.c...
[66.0%] Compiling esp32-hal-gpio.c...
[68.0%] Compiling esp32-hal-i2c.c...
[70.0%] Compiling esp32-hal-ledc.c...
[72.0%] Compiling esp32-hal-matrix.c...
[74.0%] Compiling esp32-hal-misc.c...
[76.0%] Compiling esp32-hal-psram.c...
[78.0%] Compiling esp32-hal-rmt.c...
[80.0%] Compiling esp32-hal-sigmadelta.c...
[82.0%] Compiling esp32-hal-spi.c...
[84.0%] Compiling esp32-hal-time.c...
[86.0%] Compiling esp32-hal-timer.c...
[88.0%] Compiling esp32-hal-touch.c...
[90.0%] Compiling esp32-hal-uart.c...
[92.0%] Compiling stdlib_noniso.c...
[94.0%] Compiling wiring_pulse.c...
[96.0%] Compiling wiring_shift.c...
[98.0%] Linking everything together...
[100.0%] Creating binary files...
'{tools.gen_esp32part.cmd}' is not recognized as an internal or external command,
operable program or batch file.
[Build] Error occurred.
`

H4R0, 2020-02-03:
same problem. anyone has a fix ?

Lbeloni, 2020-04-20:
Same here! Spending lots of time here and nothing !! Anyone?

ChrisBartley99, 2021-01-02:
Getting exactly the same issue ...
'{tools.gen_esp32part.cmd}' is not recognized as an internal or external command,

So near yet so far - frustrating

ttlg59, 2021-03-24:
This is a way to fix the error temporarily.
Edit "platform.txt" file, 

```
-tools.esptool_py={runtime.tools.esptool_py.path}
-tools.esptool_py.cmd=esptool
-tools.esptool_py.cmd.linux=esptool.py
-tools.esptool_py.cmd.windows=esptool.exe
+esptool_py.path={runtime.tools.esptool_py.path}
+esptool_py.cmd=esptool
+esptool_py.cmd.linux=esptool.py
+esptool_py.cmd.windows=esptool.exe

-tools.esptool_py.network_cmd=python "{runtime.platform.path}/tools/espota.py"
-tools.esptool_py.network_cmd.windows="{runtime.platform.path}/tools/espota.exe"
+esptool_py.network_cmd=python "{runtime.platform.path}/tools/espota.py"
+esptool_py.network_cmd.windows="{runtime.platform.path}/tools/espota.exe"

-tools.gen_esp32part.cmd=python "{runtime.platform.path}/tools/gen_esp32part.py"
-tools.gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"
+gen_esp32part.cmd=python "{runtime.platform.path}/tools/gen_esp32part.py"
+gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"
```
The problem at here is the Stino is filtering string key "**tools.**" to get command for Arduino upload, program and bootloader tools. So building tools use string "tools." for start string of variables will make Stino was misreaded