Robot-Will/Stino #527: Stino will download apple OSX version of avr tools on Linux Ubuntu

milosh-dev, 2020-04-10:
Fresh install of Stino on Ubuntu 18.04 machine. Added Arduino AVR platform version 1.8.2

For some reason it downloads an Apple version of avr tools. As a result build files with the "Exec format error" and file ./avr-g++ command in appropriate ~/.arduino15 folder shows that it is indeed osx format file: " Mach-O 64-bit x86_64 executable". 

I did a reinstall of .arduino15 folder at it shows that it downloads avr tools from this file:
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2

What should I do to get the correct toolchain?

GITS-SAC, 2020-04-12:
Hi, I find a solution for the problem after many effort.
Go to Arduino ==> go to Build folder and open it, go to .arduino15/packages/arduino/tools/avr-gcc/7.3.0-amtel3.6.1-arduino5 and you find the osx files.
Go to Arduino website and download the correct version : https://www.arduino.cc/en/main/OldSoftwareReleases (1.8.2 Linux)
Open the file and go to Hardware/tools/avr
Copy all the content and paste to the Build folder to replace the correct file.
Sorry if my english is bad but I hope it works for you.

GITS-SAC

milosh-dev, 2020-04-23:
I managed to install it. 

But well... the package seems to be unmantained (last commit in 2017) and it also has a memory leak issue described in [#463](https://github.com/Robot-Will/Stino/issues/463), so I won't expect it would be fixed soon in the code.